moveit/moveit #3706: How to use moveit_ros_control_interface

Cody-Vu, 2025-05-02:
Hello everyone,

I am facing an issue, where I cannot use a controller through the [moveit_ros_control_interface::MoveItControllerManager](https://github.com/moveit/moveit/blob/master/moveit_plugins/moveit_ros_control_interface/README.md). When I use it through the [SimpleControllerManager](https://github.com/moveit/moveit/blob/master/moveit_plugins/moveit_simple_controller_manager/src/moveit_simple_controller_manager.cpp) it works just fine. So the configurations are the same but while using the `moveit_ros_control_interface` I receive this error whenever I try to execute a motion:
```
[ERROR] [1746192075.954285289, 3.765000000]: Unable to identify any set of controllers that can actuate the specified joints: [ panda_joint1 panda_joint2 panda_joint3 panda_joint4 panda_joint5 panda_joint6 panda_joint7 ]
[ERROR] [1746192075.954349610, 3.765000000]: Known controllers and their joints:

[ERROR] [1746192075.954382753, 3.765000000]: Apparently trajectory initialization failed
[ INFO] [1746192076.001298231, 3.788000000]: ABORTED: CONTROL_FAILED
```

[Here](https://github.com/Cody-Vu/panda_moveit_config/blob/noetic-devel/config/ros_controllers.yaml) I configured the controllers list with the associated [parameters](https://github.com/Cody-Vu/panda_moveit_config/blob/noetic-devel/config/gazebo_controllers.yaml). And in [demo_gazebo.launch](https://github.com/Cody-Vu/panda_moveit_config/blob/noetic-devel/launch/demo_gazebo.launch#L8) I set the controller as default. 

I'd appreciate if someone could explain what I have to adjust in order to use the moveit_ros_control_interface with the cartesian impedance controller. I have to use this controller manager to switch between controllers during runtime, which is not available in the SimpleControllerManager. Thanks in advance!

rhaschke, 2025-05-02:
MoveItSimpleControllerManager uses a different format of controller specifications than ros_control!
You don't need to list anything in `ros_controllers.yaml`. The definition in `gazebo_controllers.yaml` should be sufficient.

Probably, there is some other error or warning message before, stating that the controller setup failed.

Cody-Vu, 2025-05-02:
I couldn't find any warnings regarding a controller setup fail. I also compared the output of both controller managers. Both identical.

Loading the controller seems correct:
```
[INFO] [1746192065.627161, 0.037000]: Loading controller: CartesianImpedance_trajectory_controller
[ INFO] [1746192065.667922020, 0.079000000]: Initializing Cartesian impedance controller in namespace: /CartesianImpedance_trajectory_controller
[ INFO] [1746192065.669067082, 0.080000000]: End effektor link is: panda_link8
[ INFO] [1746192065.677242871, 0.088000000]: Number of joints specified in parameters: 7
[ INFO] [1746192065.694594205, 0.107000000]: Number of joints found in urdf: 9
[ WARN] [1746192065.694640507, 0.107000000]: Number of joints in the URDF is greater than supplied number of joints: 9 > 7. Assuming that the actuated joints come first.
[ INFO] [1746192065.744608741, 0.157000000]: Finished initialization.
[INFO] [1746192065.745176, 0.158000]: Controller Spawner: Loaded controllers: franka_state_controller, CartesianImpedance_trajectory_controller
[ INFO] [1746192065.749477397, 0.162000000]: Started Cartesian Impedance Controller
[INFO] [1746192065.750294, 0.162000]: Started controllers: franka_state_controller, CartesianImpedance_trajectory_controller
```
`rosservice call /controller_manager/list_controllers` also states that the controller is running correct.

```
name: "CartesianImpedance_trajectory_controller"
    state: "running"
    type: "cartesian_impedance_controller/CartesianImpedanceController"
    claimed_resources: 
      - 
        hardware_interface: "hardware_interface::EffortJointInterface"
        resources: 
          - panda_joint1
          - panda_joint2
          - panda_joint3
          - panda_joint4
          - panda_joint5
          - panda_joint6
          - panda_joint7
```

rhaschke, 2025-05-05:
The answer can be found in the [docs](https://moveit.github.io/moveit_tutorials/doc/controller_configuration/controller_configuration_tutorial.html#ros-controllers-with-a-followjointtrajectory-action-interface):
> The MoveIt ROS Control Controller Manager will regard any controllers loaded by ROS Controller Manager as managed if it finds a plugin registration that links the type of the ROS controller with a MoveIt Controller Handle Allocator.

Hence, your new controller needs to be announced to MoveIt. Thus, create a plugin description file:

```xml
<library path="libmoveit_ros_control_interface_trajectory_plugin">
    <class
        name="my_controller_package_name/my_controller_type_name"
        type="moveit_ros_control_interface::JointTrajectoryControllerAllocator"
        base_class_type="moveit_ros_control_interface::ControllerHandleAllocator">
    </class>
</library>
```

and reference this plugin description file in your package.xml’s export section:

```xml
<exec_depend>moveit_ros_control_interface</exec_depend>
<export>
    <moveit_ros_control_interface plugin="${prefix}/moveit_controller_plugin.xml"/>
</export>
```

Cody-Vu, 2025-05-05:
I refeferenced the [plugin description file](https://github.com/Cody-Vu/Cartesian-Impedance-Controller/blob/master/cartesian_impedance_moveit_plugin.xml) in the [package.xml](https://github.com/Cody-Vu/Cartesian-Impedance-Controller/blob/master/package.xml#L55)’s export section. 

> After building the package, any controllers in ros_controllers.yaml that reference controller_package_name/controller_type_name will become available for use with MoveIt.

According to the docs I added the controller to [ros_controllers.yaml](https://github.com/Cody-Vu/panda_moveit_config/blob/noetic-devel/config/ros_controllers.yaml), but also tried to let it empty as you mentioned it isn't necessary. Both approaches ended in the same error again (Unable to identify any set of controllers...). Can you have a look at my plugin setup by any chance?

rhaschke, 2025-05-05:
What is the output of `rospack plugins --attrib=plugin moveit_ros_control_interface`?
It should list the newly added plugin.xml file.

rhaschke, 2025-05-05:
It is fine to let ros_controllers.yaml empty.

rhaschke, 2025-05-05:
The culprit is the [`name` in your plugin.xml file](https://github.com/Cody-Vu/Cartesian-Impedance-Controller/blob/master/cartesian_impedance_moveit_plugin.xml#L3): it should match the `type` in your gazebo_controllers.yaml:
```xml
<library path="libmoveit_ros_control_interface_trajectory_plugin">
  <class name="cartesian_impedance_controller/CartesianImpedanceController"
    type="moveit_ros_control_interface::JointTrajectoryControllerAllocator"
    base_class_type="moveit_ros_control_interface::ControllerHandleAllocator">
    <description></description>
  </class>
</library>
```

Cody-Vu, 2025-05-05:
Here is the output:
```
cartesian_impedance_controller /home/catkin_ws/src/Cartesian-Impedance-Controller/cartesian_impedance_moveit_plugin.xml
moveit_ros_control_interface /home/scrubnurse/catkin_test/src/moveit/moveit_plugins/moveit_ros_control_interface/moveit_ros_control_interface_plugins.xml
cartesian_impedance_controller /home/scrubnurse/catkin_test/src/Cartesian-Impedance-Controller/cartesian_impedance_moveit_plugin.xml
moveit_ros_control_interface /home/scrubnurse/catkin_test/src/moveit/moveit_plugins/moveit_ros_control_interface/moveit_ros_control_interface_plugins.xml
```

Cody-Vu, 2025-05-05:
> <library path="libmoveit_ros_control_interface_trajectory_plugin">
>   <class name="cartesian_impedance_controller/CartesianImpedanceController" [...]

Thank you very much for your kind help and clarification @rhaschke ! You are right, matching the type like in the gazebo_conrollers.yaml was the soultion. I really appreciate your time and support!

closed 2025-05-05 by Cody-Vu