moveit/moveit #3708: Can't add moveit interactive marker due to Namespace

crazy-robotics, 2025-05-06:
I am using Moveit for robotic arm simulation. I want to use multiple robotic arms in Gazebo simulation. To differentiate the `topic` and `services`  I use ros namespace technique. here is my some of servieces list
```
/arm1/apply_planning_scene
/arm1/get_planner_params
/arm1/get_planning_scene
/arm1/move_group/plan_execution/set_parameters
/arm1/move_group/planning_scene_monitor/set_parameters
/arm1/move_group/sense_for_plan/set_parameters
/arm1/plan_kinematic_path
/arm1/query_planner_interface
/arm1/set_planner_params
/rviz_1746536378494151947/motionplanning_planning_scene_monitor/set_parameters
``` 
when I open `rviz` and add `MotionPlanning` I got following error in my terminal 

```
waitForService: Service [/get_planning_scene] has not been advertised, waiting...

```
due to which I can not see or add `moveit interactive marker` in `MotionPlanning`
I have the service `/arm1/get_planning_scene`. but `moveit` wait for `[/get_planning_scene]` I don't know to change it

rhaschke, 2025-05-06:
Did you try setting the field `Move Group Namespace`?

[image]

crazy-robotics, 2025-05-07:
when I check services list `rosservice list | grep plan`
```
/arm1/apply_planning_scene
/arm1/get_planner_params
/arm1/get_planning_scene
/arm1/move_group/plan_execution/set_parameters
/arm1/move_group/planning_scene_monitor/set_parameters
/arm1/move_group/sense_for_plan/set_parameters
/arm1/plan_kinematic_path
/arm1/query_planner_interface
/arm1/set_planner_params
/rviz_1746602956347401207/motionplanning_planning_scene_monitor/set_parameters
```
I have `/arm1/apply_planning_scene` I also get following terminal error from where rviz is open 
```
[ INFO] [1746603128.036355080]: waitForService: Service [/get_planning_scene] has not been advertised, waiting...
[ INFO] [1746603133.060803321]: Failed to call service get_planning_scene, have you launched move_group or called psm.providePlanningSceneService()?

```
it is listening to `[/get_planning_scene] ` but I have `/arm1/get_planning_scene` service

rhaschke, 2025-05-07:
Did you try setting the field `Move Group Namespace` in rviz as suggested in https://github.com/moveit/moveit/issues/3708#issuecomment-2855280160?

crazy-robotics, 2025-05-07:
I changed the Move Group Namespace to as you mentioned above I tried `/arm1` and `/arm1/get_planning_scene` getting the same error 
```
[ INFO] [1746605798.488941288]: Listening to '/arm1/get_planning_scene/planning_scene'
[ INFO] [1746605798.490025306]: waitForService: Service [/arm1/get_planning_scene/get_planning_scene] has not been advertised, waiting...
[ INFO] [1746605803.498343680]: Failed to call service /arm1/get_planning_scene/get_planning_scene, have you launched move_group or called psm.providePlanningSceneService()?
```

rhaschke, 2025-05-07:
Obviously, you need to use the `arm1` or `/arm1`.

crazy-robotics, 2025-05-07:
I can see `OMPL`, but there is no `interactivemarker` to move the ` End Effector` 

[image]

rhaschke, 2025-05-07:
Looks like you solved the initial issue. If a marker is still not shown (although having no warnings on the console anymore?), you should check with a simple single-robot setup first: Is a marker shown there?
If you have a non-trivial kinematic tree with multiple leaves, you need to define an end-effector for your joint groups.